google-deepmind/mujoco_mpc #375: Subspace Planning

chongwilliam, 2025-03-12:
Hi @thowell,

I wanted to ask about a feature addition into the planners, which is to plan within a subspace of the robot's control space.  The subspace would be defined by a basis V, with the orthogonal projection matrix VV^{T}.  I would be mainly interested in the derivative-based approaches for this purpose.

Thanks!

Tadinu, 2025-03-12:
@chongwilliam may it be related to #370

thowell, 2025-03-12:
@yuvaltassa do you think a plugin would make sense for this type of application?

yuvaltassa, 2025-03-15:
The way I would do this depends on the answer to two questions. Let's call the size of your subspace `nw`, so the matrix $V$ is `nu x nw`.

1. Is `nw` fixed or varying? If varying, would it vary between _episodes_, _planner iterations_ or _timesteps within a rollout_? Obviously these three options correspond to "easy", "hard" and "forget about it" 🙂  
2. Now assuming `nw` is fixed or mostly fixed, how often do you imagine $V$ itself varying?

chongwilliam, 2025-03-15:
In my application, I want to impose motion constraints given by geometric constraints between a manipulated object and the environment, for instance. Here, nw would vary based on the current interaction, so I would assume this means per planner iteration?  Similarly, V would vary based on the interaction (maybe every few seconds or so).

yuvaltassa, 2025-03-16:
Oh wow. This does not sound like a good idea at all. If you like, I can explain why this is a bad idea and/or provide examples of ideas that I think are good.

To do that I need to understand if you are
1. Interested in subspace planning and looking for an application.
2. Interested in imposing motion constraints and think that subspace planning might be a good solution.
3. Interested in this specific combination of problem and solution approach.

chongwilliam, 2025-03-18:
I'm interested in 2.  My goal is to compute online path modifications based on state estimation (object poses, force information, ...) for a nominal motion plan, where the modifications exist in a subspace.  For example, in assembly tasks, there might be instances of jamming, where some constrained perturbations can be performed to un-jam the manipulated piece.  I'm interested in hearing about your thoughts about the applicability of Mujoco MPC for this type of application.