google-deepmind/mujoco_mpc #378: Build issue - c++: error: unsupported option '-msse4.1' for target 'arm64-apple-darwin24.3.0'

b-alfaro, 2025-04-21:
I am trying to build on an M3 Mac. I run  `cmake ..` and `make -j`. After the second command, I see this error
c++: error: unsupported option '-msse4.1' for target 'arm64-apple-darwin24.3.0'. Has anyone ran into this or found a workaround?

dmarew, 2025-05-17:
@b-alfaro , I ran into the same build error and got it working by pinning abseil-cpp to a newer commit in the CMake file. Just drop this into your CMakeLists.txt 
`set(MUJOCO_DEP_VERSION_abseil
    # fb3621f4f897824c0dbe0615fa94543df6192f30  # LTS 20230802.1 (old, broken)
    bc257a88f7c1939f24e0379f14a3589e926c950c  # updated working commit
    CACHE STRING "Version of `abseil` to be fetched."
    FORCE
)`

sprakashdash, 2025-06-12:
I am following the installation step by step but when doing cmake used the following command: `cmake .. -DCMAKE_BUILD_TYPE:STRING=Release -G Ninja -DMJPC_BUILD_GRPC_SERVICE:BOOL=ON -DCMAKE_POLICY_VERSION_MINIMUM=3.5 -Wno-dev`
I have also changed the `MUJOCO_DEP_VERSION_abseil ` but still getting the following error:

```
(mjpc) uuser@E-LOSX3J72G build % cmake --build . --config=Release
[0/2] Re-checking globbed directories...
[3/1890] Building CXX object _deps/grpc-build/CMakeFiles/grpc.dir/src/core/ext/filters/client_channel/client_channel_factory.cc.o
FAILED: _deps/grpc-build/CMakeFiles/grpc.dir/src/core/ext/filters/client_channel/client_channel_factory.cc.o 
/usr/bin/c++ -DCARES_STATICLIB -DSOURCE_DIR=\"/Users/user/Documents/mujoco_mpc\" -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/third_party/re2 -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/include -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/third_party/address_sorting/include -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-build/third_party/re2 -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/third_party/boringssl-with-bazel/src/include -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/src/core/ext/upb-gen -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/src/core/ext/upbdefs-gen -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/third_party/upb -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/third_party/utf8_range -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/third_party/xxhash -I/Users/user/Documents/mujoco_mpc/build/_deps/abseil-cpp-src -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-build/third_party/cares/cares -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/third_party/cares/cares -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/third_party/cares/cares/include -O3 -DNDEBUG -std=c++20 -flto=thin -arch arm64 -fPIC -fvisibility=hidden -fvisibility-inlines-hidden -fdata-sections -ffunction-sections -MD -MT _deps/grpc-build/CMakeFiles/grpc.dir/src/core/ext/filters/client_channel/client_channel_factory.cc.o -MF _deps/grpc-build/CMakeFiles/grpc.dir/src/core/ext/filters/client_channel/client_channel_factory.cc.o.d -o _deps/grpc-build/CMakeFiles/grpc.dir/src/core/ext/filters/client_channel/client_channel_factory.cc.o -c /Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/src/core/ext/filters/client_channel/client_channel_factory.cc
In file included from /Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/src/core/ext/filters/client_channel/client_channel_factory.cc:21:
In file included from /Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/src/core/ext/filters/client_channel/client_channel_factory.h:24:
In file included from /Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/src/core/ext/filters/client_channel/subchannel.h:35:
In file included from /Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/src/core/ext/filters/client_channel/connector.h:31:
In file included from /Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/src/core/lib/transport/transport.h:58:
In file included from /Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/src/core/lib/promise/pipe.h:41:
In file included from /Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/src/core/lib/promise/seq.h:24:
/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/src/core/lib/promise/detail/basic_seq.h:103:38: error: a template argument list is expected after a name prefixed by the template keyword [-Wmissing-template-arg-list-after-template-kw]
  103 |                     Traits::template CallSeqFactory(f_, *cur_, std::move(arg)));
      |                                      ^
1 error generated.
[4/1890] Building CXX object _deps/grpc-build/CMakeFiles/grpc.dir/src/core/ext/filters/client_channel/dynamic_filters.cc.o
FAILED: _deps/grpc-build/CMakeFiles/grpc.dir/src/core/ext/filters/client_channel/dynamic_filters.cc.o 
/usr/bin/c++ -DCARES_STATICLIB -DSOURCE_DIR=\"/Users/user/Documents/mujoco_mpc\" -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/third_party/re2 -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/include -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/third_party/address_sorting/include -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-build/third_party/re2 -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/third_party/boringssl-with-bazel/src/include -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/src/core/ext/upb-gen -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/src/core/ext/upbdefs-gen -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/third_party/upb -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/third_party/utf8_range -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/third_party/xxhash -I/Users/user/Documents/mujoco_mpc/build/_deps/abseil-cpp-src -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-build/third_party/cares/cares -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/third_party/cares/cares -I/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/third_party/cares/cares/include -O3 -DNDEBUG -std=c++20 -flto=thin -arch arm64 -fPIC -fvisibility=hidden -fvisibility-inlines-hidden -fdata-sections -ffunction-sections -MD -MT _deps/grpc-build/CMakeFiles/grpc.dir/src/core/ext/filters/client_channel/dynamic_filters.cc.o -MF _deps/grpc-build/CMakeFiles/grpc.dir/src/core/ex [... cut ...]
In file included from /Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/src/core/lib/promise/seq.h:24:
/Users/user/Documents/mujoco_mpc/build/_deps/grpc-src/src/core/lib/promise/detail/basic_seq.h:103:38: error: a template argument list is expected after a name prefixed by the template keyword [-Wmissing-template-arg-list-after-template-kw]
  103 |                     Traits::template CallSeqFactory(f_, *cur_, std::move(arg)));
      |                                      ^
1 error generated.
ninja: build stopped: subcommand failed.
```

amiao1017, 2025-07-31:
> [@b-alfaro](https://github.com/b-alfaro) , I ran into the same build error and got it working by pinning abseil-cpp to a newer commit in the CMake file. Just drop this into your CMakeLists.txt [...]

NOT WORK FOR MAC MINI M4

joaocandre, 2025-08-12:
@amiao1017 I also get this error when building gRPC, so it should be an upstream issue. I builds successfully if you remove `-DMJPC_BUILD_GRPC_SERVICE:BOOL=ON`.